Pedro-Pathing/PedroPathing #20: Telemetry Manager

Logan-Nash, 2025-04-29:
- [ ] Utilizing Log.d for telemetry feedback (easier to get robot logs n other stuff thanks QC for recommendation)
- [ ] Creation of telemetry parent and child class
- [ ] Refractor codebase to properly use new classes

BeepBot99, 2025-04-29:
I'd say have `Logger`, with multiple implementations (including `FTCTelemetryLogger`). Also have a `CombinedLogger` that notifies all of its child loggers.
Alternatively, use the publisher-subscriber pattern.

BaronClaps, 2025-04-30:
> I'd say have `Logger`, with multiple implementations (including `FTCTelemetryLogger`). Also have a `CombinedLogger` that notifies all of its child loggers. Alternatively, use the publisher-subscriber pattern.

This was the idea 👍🏻

BeepBot99, 2025-04-30:
> This was the idea 👍🏻

Alternatively, use the publisher-subscriber pattern (which might be better):

```java
@FunctionalInterface
public interface LogSubscriber {
  void onLog(String key, Object value);
  default void update() { }
}
```

```java
public final class Logger {
  private static final List<LogSubscriber> subscribers = new ArrayList<>();

  public static void addSubscriber(LogSubscriber subscriber) {
    subscribers.add(subscriber);
  }

  public static void log(String key, Object value) {
    for (LogSubscriber subscriber : subscribers) {
      subscriber.onLog(key, value);
    }
  }

  public static void update() {
    for (LogSubscriber subscriber : subscribers) {
      subscriber.update();
    }
  }

  public static void clearSubscribers() {
    subscribers.clear();
  }
}
```

There would be an implementation for `TelemetryLogger`, `PanelsLogger`, etc. You can also use a lambda, if desired. Since all methods are static, classes can just log with `Logger.log(key, value)`.

BaronClaps, 2025-05-01:
> > This was the idea 👍🏻
> 
> Alternatively, use the publish-subscribe pattern (which might be better):
> 
> @FunctionalInterface
> public interface LogSubscriber {
>   void onLog(String key, Object value);
>   default void update() { }
> }
> public final class Logger {
>   private static final List<LogSubscriber> subscribers = new ArrayList<>();
> 
>   public static void addSubscriber(LogSubscriber subscriber) {
>     subscribers.add(subscriber);
>   }
> 
>   public static void log(String key, Object value) {
>     for (LogSubscriber subscriber : subscribers) {
>       subscriber.onLog(key, value);
>     }
>   }
> 
>   public static void update() {
>     for (LogSubscriber subscriber : subscribers) {
>       subscriber.update();
>     }
>   }
> 
>   public static void clearSubscribers() {
>     subscribers.clear();
>   }
> }
> There would be an implementation of `LogSubscriber` for `TelemetryLogger`, `PanelsLogger`, etc. You can also use a lambda, if desired. Since all methods are static, classes can just log with `Logger.log(key, value)`.

This is a cool way of doing this. I do like this.

BaronClaps, 2025-06-27:
Panels + other plans may handle this already. Revisit later if needed.

BeepBot99, 2025-07-08:
What are you going to do?

BaronClaps, 2025-07-08:
> What are you going to do?

A publish-subscribe pattern + log exporting in a pedro logging format + wpilib logging conversion