Pedro-Pathing/PedroPathing #22: Exploring FTC-Panels and enabling compatibility

Logan-Nash, 2025-04-29:
- [ ] Figure out ftc panels and ensure our userbase can use FTC Panels with Pedro Pathing
- [ ] Ensure dual compatibility with FTCDash and FTC Panels

BeepBot99, 2025-04-29:
I think the best way to go about doing this would be to not have a dependency on either the dashboard or panels. 
If constants are no longer in the global `FollowerConstants`, then `FollowerConstants` no longer needs to be annotated with `@Config`. I'm pretty sure the only other place the dashboard is used is in `Follower#telemetryDebug`, which takes a `MultipleTelemetry`. I know that `MultipleTelemetry` implements `Telemetry`, so probably just make it take `Telemetry` instead. (Also logging will be in a different class.)
Then for drawing on the dashboard, I don't know how that works and I doubt it's possible with `Telemetry`. Possibly another module that has a dependency on the dashboard for drawing on it?
And as for panels, I don't know if it has support for drawing on a field. If it does, maybe have a module for panels as well?
Alternatively, if there is only one class for dashboard and one class for panels, they can just be in the main module with a `compileOnly` dependency on each of them, if you're careful.

BaronClaps, 2025-04-29:
> I think the best way to go about doing this would be to not have a dependency on either the dashboard or panels. 
> Alternatively, if there is only one class for dashboard and one class for panels, they can just be in the main module with a `compileOnly` dependency on each of them, if you're careful.

I believe both of these ideas were the goal - using Log.d as the default logging method / feedback