Pedro-Pathing/PedroPathing #23: Interface based Follower

Logan-Nash, 2025-04-29:
- [ ] Drivetrain Interface
- [ ] Localizer Interface
- [ ] Feedback Manager Interface (ties with telemetry?)
Maybe others but still working on the possibilities

BeepBot99, 2025-04-29:
[image]
What about something like this? (An arrow represents a dependency, and diamonds are interfaces and rectangles are classes.)

- Very testable, as everything depends on interfaces only
  - This also means it follows the Dependency Inversion Principle
- Logger can have different implementations, such as one for the FTC SDK telemetry and another for WPILib telemetry.
  - I'd prefer Logger as it is the standard name. FeedbackManager could get confused with feedback controllers.
- PoseUpdater is renamed to PoseTracker. This is entirely optional, but I think it better conveys the purpose.
- DrivePowerCalculator is responsible for calculating the drive powers, given the current pose and path. 
- VectorDrivePowerCalculator is Pedro's algorithm. However, it could also be easily swapped out for another algorithm, if needed.
- VectorCalculator calculates the four vectors. The default implementation is PIDVectorCalculator, which uses PIDs. 
  - There could be one implementation for single PIDs and another implementation for large and small PIDs.
  - There could also be an implementation for SquID.
  - Since the centripetal vector does not rely on PIDs, maybe VectorDrivePowerCalculator would have four VectorCalculators, and the default would be to use three PIDVectorCalculators and one CentripetalVectorCalculator? Although if VectorDrivePowerCalculator always has translational, heading, drive, and centripetal vectors, there isn't really another option for centripetal. Maybe VectorDrivePowerCalculator should just do centripetal then?
- VectorCombiner combines the four vectors into drive powers. DefaultVectorCombiner would be what Pedro does currently. UniformVectorCombiner would weigh them all equally. I believe this was asked for at some point?

BeepBot99, 2025-04-29:
That sounds good. But why should drivetrain create the motor powers from that? Wouldn't that be the same regardless of drivetrain? Maybe that should be extracted into `VectorCombiner`.

BaronClaps, 2025-04-29:
> That sounds good. But why should drivetrain create the motor powers from that? Wouldn't that be the same regardless of drivetrain? Maybe that should be extracted into `VectorCombiner`.

Because it will depend between different drivetrains what the motors will be powered (ex. swerve will use servos + motors)

BeepBot99, 2025-04-29:
Yeah but I mean like are you saying Drivetrain should take the four vectors and combine them into drive powers, or am I did I just misunderstand? Because if that's the case, I believe that would be the same regardless of drivetrain, where drive powers isn't the actual motor powers but is a vector consisting of x, y, and heading (could probably be stored in a Pose)

AnyiLin, 2025-04-29:
I think what barons saying is that we can take in the errors and just calculate all the correctional vectors from there, which I think is pretty reasonable since this is all PID run so error and PID coefficients is all we need to calculate most of the vectors. For centripetal, we just pass in the paths second derivative at the point instead of error. Then, Drivetrain will take in these vectors and translate them into motor powers.

BeepBot99, 2025-04-29:
> I think what barons saying is that we can take in the errors and just calculate all the correctional vectors from there, which I think is pretty reasonable since this is all PID run so error and PID [...]

OK. Let's say we have two drivetrains: mecanum and swerve. If I understand correctly, there are two steps:

1. Convert the four vectors into motor powers.
2. Set the motor powers to those calculated motor powers.

However, the first step would (I think) be largely duplicated between the two drivetrains. IIRC Pedro weighs the four vectors somehow, and that would be duplicated. Additionally, one might desire to weigh them differently. I think we could have:

```java
@FunctionalInterface
public interface VectorCombiner {
  Pose combineVectors(Vector translational, Vector heading, Vector drive, Vector centripetal);
}

@FunctionalInterface
public interface Drivetrain {
  void setMotorPowers(Pose movementVector);
}
```

BaronClaps, 2025-04-30:
> 1. Convert the four vectors into motor powers.
> 2. Set the motor powers to those calculated motor powers.
>
> However, the first step would (I think) be largely duplicated between the two drivetrains. [...]

We feel that the "VectorCombiner" that you are proposing wouldn't be of much use. Users can create new Drivetrains within their project if they wish to change the weighting. We aren't totally set on this, but it does simplify for, what I think, is a very small use case.

BeepBot99, 2025-04-30:
> I feel that the "VectorCombiner" that you are proposing wouldn't be of much use. Users can create new Drivetrains within their project if they wish to change the weighting. We aren't totally set on this, but it does simplify for, what I think, is a very small use case.

Hmm, that may be true. Alright then, however I still would prefer that the default drivetrains are in the library, and users can just copy and paste it into their code if changes are needed (and very easily).

Also, Drivetrain and Localizer should be interfaces, not abstract classes, for two reasons:

1. Abstract classes have MUCH more runtime overhead than interfaces. Interfaces have zero runtime overhead. Abstract classes require memory to be allocated for an object every time an instance is created.
2. Drivetrain and Localizer define contracts for classes, which is what interfaces are for.

BaronClaps, 2025-05-01:
I made Localizer an interface, but I feel Drivetrain might be better as an abstract.

BeepBot99, 2025-05-01:
> I made Localizer an interface, but I feel Drivetrain might be better as an abstract.

Why is that?

BaronClaps, 2025-05-01:
It feels like it works better as there would be a max power scaling no matter what, but I guess it could become interface.

[image]

BeepBot99, 2025-05-01:
> It feels like it works better as there would be a max power scaling no matter what, but I guess it could become interface.

Oh hmm, maybe then. Is there a reason that `getDrivePowers` and `runPowers` need to be separate? Maybe for power scaling or something?
Possibly (I know we've already talked about this :sob: but it could help) splitting it into `Drivetrain` and `VectorCombiner` could help with this. A drivetrain could take a vector with x, y, and heading powers, and it could be agnostic of whether it is in auto or TeleOp. (It could be stored in a Pose.)

Then `Drivetrain` could look as follows:

```java
public abstract class Drivetrain {
  private double maxPowerScaling;

  public void setMaxPowerScaling(double maxPowerScaling) {
    this.maxPowerScaling = MathFunctions.clamp(maxPowerScaling, 0, 1);
  }

  public double getMaxPowerScaling() {
    return maxPowerScaling;
  }

  public void setUnscaledDrivePowers(Pose drivePowers) {
    setDrivePowers(drivePowers.times(maxPowerScaling));
  }
  
  public abstract void setDrivePowers(Pose drivePowers);
}
```
And then `VectorCombiner` could be:

```java
public interface VectorCombiner {
  Pose getDrivePowers(Vector correctivePower, Vector headingPower, Vector pathingPower, double robotHeading);
}
```

With this `Drivetrain`, it is also agnostic of auto vs TeleOp. In TeleOp, follower can just pass the joystick inputs!

BaronClaps, 2025-05-01:
> Oh hmm, maybe then. Is there a reason that `getDrivePowers` and `runPowers` need to be separate? Maybe for power scaling or something? 
There currently isn't a use case for this, but I could see someone wanting to pull the drivePowers. THere ocul dbe a combine method that could run both.

BeepBot99, 2025-05-01:
There currently isn't a use case for this, but I could see someone wanting to pull the drivePowers. THere ocul dbe a combine method that could run both.

Sure, although they'd likely end up calling it twice (because the Follower already calls it once), which I assume is undesirable. If this is truly needed, maybe a cached value or publish-subscribe pattern would be better here?

BaronClaps, 2025-05-01:
This should mostly be done - other than the Feedback manager. A few things need to happen with VectorCalculator to make the PIDFCoefficents accessible during tuning, but otherwise the Follower seems very close to being done. Around 500 lines in the follower now.

BaronClaps, 2025-05-02:
Yeah, I am fine with that, I was referring to it as the name in the issue description.

BaronClaps, 2025-05-02:
Done.

closed 2025-05-02 by BaronClaps