Pedro-Pathing/PedroPathing #24: Constants

Logan-Nash, 2025-04-29:
Because our constants handling system sucked before, here's what we want to improve:

- [ ] Constants is an Object passed into the follower

- [ ] Constants are not drivetrain specific

- [ ] Allocates for general settings (constraints)

- [ ] Able to add drivetrain constants

- [ ] Avoiding Localization constants if possible

BeepBot99, 2025-04-29:
I think the best way to do this would be to have each class have its own constants, possibly even a nested class for less files. No class should have access to constants other than its own. 
I'd actually prefer the opposite of what you say, where everything has constants. For example,

- `MecanumDrivetrain` has `MecanumDrivetrainConstants` (or `MecanumDrivetrain.Constants`)
- `ThreeWheelLocalizer` has `ThreeWheelLocalizerConstants`
- And so on

This way,

a. Nothing has access to more constants than it needs.
b. The user isn't forced to provide any constants they don't use.

BaronClaps, 2025-04-29:
I believe this was the intent as well, it would just have to get passed down a chain

BeepBot99, 2025-04-29:
> I believe this was the intent as well, it would just have to get passed down a chain

What do you mean passed down a chain?

AnyiLin, 2025-04-29:
Moving the path constraint constants into the path classes or something similar would also be good

BaronClaps, 2025-04-29:
> Moving the path constraint constants into the path classes or something similar would also be good

Along with this, better defaults can be implemented on the library side, then users can just change constraints on a path to path basis.

BaronClaps, 2025-04-29:
> > I believe this was the intent as well, it would just have to get passed down a chain
> 
> What do you mean passed down a chain?

Passing the constants through the follower to the various sub classes

BeepBot99, 2025-04-29:
I don't like that, because that requires follower to:

a. Know the implementations of its dependencies (it should only know of the interfaces)
b. Construct its dependencies (bad for the same reason)

I'd say follower should only take its dependencies. As for making it easier to construct them all, there can be a follower builder, which would be the only place those "rules" are broken.

BaronClaps, 2025-05-01:
Currently working on this. I am wondering if it would be smart to have the constants continuely update and are changable within dash, or handle the changing of, for example, the PIDFCoefficents in the vector manager. I feel within the Constants will provide a centralized approach. We could make it a regular class so we can define variables and methods to pull those variables. Then users can create child classes and only have to change the constants they want to in the construtor? @BeepBot99 @Logan-Nash what do you think?

BeepBot99, 2025-05-02:
> Currently working on this. I am wondering if it would be smart to have the constants continually update (this would be pretty simple to implement) and are changeable within dash, or handle the changing of, for example, the PIDFCoefficents in the vector manager. I feel within the Constants will provide a centralized approach. We could make it a regular class so we can define variables and methods to pull those variables. Then users can create child classes and only have to change the constants they want to in the constructor? 

Don't do things like this:

```java
translationalPIDFFeedForward = constants.translationalPIDFFeedForward();
secondaryTranslationalPIDFFeedForward = constants.secondaryTranslationalPIDFFeedForward();
useSecondaryDrivePID = constants.useSecondaryDrivePIDF();
useSecondaryHeadingPID = constants.useSecondaryHeadingPIDF();
useSecondaryTranslationalPID = constants.useSecondaryTranslationalPIDF();
```

Instead, everywhere in the class where it is used, access it directly from `constants`. That way, if it changes in `constants`, it will be updated immediately. 

Additionally, that approach for constants is one that I thought of (probably would use an IIB instead of a constructor though). However, I think *not* subclassing the constants and instead just creating an instance is better, as it leads to:

1. A lot less code duplication
2. Less unnecessary classes

It also allows you to avoid creating a separate instance of the class.

You can have something like this: (simplified)

```java
@FunctionalInterface
public interface Configurator<T> {
  void configure(T constants);
}

public final class FollowerConstants {
  public boolean useSecondaryDrivePIDF = false;
  public boolean useSecondaryTranslationalPIDF = false;

  public FollowerConstants(Configurator<FollowerConstants> configurator) {
    configurator.configure(this);
  }
}
```

Then, if `Follower` takes `FollowerConstants` and stores it in a field `constants`, it can just reference `constants.useSecondaryDrivePIDF` directly anywhere it wants to use it, so when `useSecondaryDrivePIDF` is updated it will automatically update.

Additionally, this approach allows some constants to be required instead of having a default value, if needed (this could be useful for motor names, etc.)

BaronClaps, 2025-05-02:
It should be finished. Updated the descriptions.

closed 2025-05-02 by BaronClaps