Pedro-Pathing/PedroPathing #25: Explore Choreo modified for Pedro

BaronClaps, 2025-04-29:
Potentially implement a modified version of Choreo for FTC

Logan-Nash, 2025-04-29:
I think there is a path planner port for ftc already out somewhere. I think 10179 Tech Turtles had one for their swerve and could be worth looking into

BaronClaps, 2025-04-29:
> I think there is a path planner port for ftc already out somewhere. I think 10179 Tech Turtles had one for their swerve and could be worth looking into

👍🏻

BaronClaps, 2025-04-29:
@vintheruler1

Logan-Nash, 2025-04-29:
https://github.com/Tech-Turtles/CenterStage/tree/main/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/utility/pathplanner

This is the implementation of path planner we can look into

BaronClaps, 2025-04-29:
> It also supports adding commands within the auto maker. 
At least for now, let’s focus on the Path maker side of things as we do not have an action system. 

We will need to find a way to convert the PathPlanner paths (however they are stored) into our curves + interp.

BeepBot99, 2025-04-29:
I don't know much about PathPlanner, but I don't think it would be compatible with Pedro currently. First of all, it uses motion profiling, although I don't know if it does that when it generates the paths in the app or when it is run on the robot.
But also, it has features that Pedro doesn't (currently) such as a heading interpolation that always faces a certain point while moving along a path.
Additionally, I know the PathPlanner team is working on making it work for FTC (not with the current SDK, but they're going to add support for FTC fields and such by 2027).

BaronClaps, 2025-04-29:
> I don't know much about PathPlanner, but I don't think it would be compatible with Pedro currently. First of all, it uses motion profiling, although I don't know if it does that when it generates [...]

We are focused on this for 2025-2027 before the new control system.

BeepBot99, 2025-04-29:
> We are focused on this for 2025-2027 before the new control system.

Yeah I know. Just wanted to make sure you knew.

> Path Planner does have motion profiling but it is not active when generating the paths on the app. The app simply returns a JSON file with the path, coordinates, and other information but most of the other is not useful to Pedro.

Oh OK. How are you planning on dealing with PathPlanner features that Pedro does not have, such as velocity constraints and rotation targets? The three options I see are:

- Completely ignoring them
- Forking PathPlanner and removing those options
- Adding that functionality to Pedro

Logan-Nash, 2025-04-29:
I think we'll likely go with option 2. I'm not too worried since we have people that have experience with path planner and I think it'll help with changing things around.

vintheruler1, 2025-04-29:
As Logan said, we will go with option 2. The fork has been forked and we will work on removing/customizing velocity constraints, rotation targets, current limits, PPLib, and many more FRC specific things.

Logan-Nash, 2025-04-29:
> The fork has been forked

Me when the fork has been forked

BeepBot99, 2025-04-30:
> I think we'll likely go with option 2. I'm not too worried since we have people that have experience with path planner and I think it'll help with changing things around.

Ah OK. Is this intended to replace the Pedro visualizer?

vintheruler1, 2025-04-30:
> Ah OK. Is this intended to replace the Pedro visualizer?

Yes it is intended to replace the visualizer.

I also began work on the fork today. I was able to replace the PathPlanner logos with the Pedro one and rename the it from PathPlanner to PedroPlanner. I was not able to do much apart from that but I dug deeper and found it is all organized. All of the files that need to be edited are in the **./lib** folder and in the **path** folder there is a file called **pathplanner_path.dart** where it holds most of the FRC specific data types.

On top of this, make sure you read the README.md and install Flutter to deploy the app.

BaronClaps, 2025-05-01:
@vintheruler1 Any other updates?

vintheruler1, 2025-05-04:
Hey, I have not been able to work on the port due to some exams this week but on this Thursday/Friday I will be able to continue working on the port! Sorry about the delay.