Pedro-Pathing/PedroPathing #27: Drivetrains

BaronClaps, 2025-04-30:
Move Drivetrains into the quickstart so users can modify and add custom Drivetrains, also avoids the need for most of the constants in FConstants

BeepBot99, 2025-04-30:
Personally, I don't like this (along with #26), although I know that many people do.
The main reason is because it makes it a lot harder for Pedro to make updates to the library and for users to pull those updates without possibly breaking their existing code. If it's part of the library, it should be in the library.

BeepBot99, 2025-04-30:
Adding on to what I said yesterday, putting drivetrains and localizers in the quickstart would make it more cumbersome for users to use their drivetrain. 

If the default drivetrains were in the library, there could be a fluent builder for creating the follower:

```java
follower = Follower.builder(hardwareMap, followerConstants)
    .threeWheelLocalizer(localizerConstants)
    .mecanumDrivetrain(drivetrainConstants)
    .build();
```

However, if they are in the quickstart, the user would need to pass instances manually, with `new MecanumDrivetrain()` and `new ThreeWheelLocalizer()`.

Furthermore, constants should be tied to an instance of the class, not the class itself.

> Encapsulate what varies

Additionally, the user shouldn't be required to (or have by default) *all* the implementations (such as mecanum **and** swerve) in their code, only the ones they use. Doing so violates the Interface Segregation Principle.

Lastly, as I said yesterday, having them in the quickstart makes it significantly harder for users to update their code when there are updates to them.

I also see no case where having them in the library makes it *less* customizable.

Having them in the quickstart seems to me like the quick and easy way out, instead of the best option.

BaronClaps, 2025-04-30:
> If the default drivetrains were in the library, there could be a fluent builder for creating the follower:
> 
> follower = Follower.builder(hardwareMap, followerConstants)
>     .threeWheelLocalizer(localizerConstants)
>     .mecanumDrivetrain(drivetrainConstants)
>     .build();

I do like this idea. We could also have a `.addDrivetrain(Drivetrain drivetrain, Constants constants)`, though how do we define the `drivetrainConstants`? Would they be a child class of `Constants`? Same thing goes for localizers.

BeepBot99, 2025-04-30:
> I do like this idea. We could also have a `.addDrivetrain(Drivetrain drivetrain, Constants constants)`, though how do we define the `drivetrainConstants`? Would they be a child class of `Constants`? Same thing goes for localizers.

Drivetrains could take their constants as a constructor parameter. So there could be:

```java
FollowerBuilder addDrivetrain(Drivetrain drivetrain);

FollowerBuilder addMecanumDrivetrain(MecanumDrivetrainConstants constants) {
  return addDrivetrain(new MecanumDrivetrain(constants));
}
```

BaronClaps, 2025-04-30:
> > I do like this idea. We could also have a `.addDrivetrain(Drivetrain drivetrain, Constants constants)`, though how do we define the `drivetrainConstants`? Would they be a child class of `Constants`? Same thing goes for localizers.
> 
> Drivetrains could take their constants as a constructor parameter. So there could be:
> 
> FollowerBuilder addDrivetrain(Drivetrain drivetrain);
> 
> FollowerBuilder addMecanumDrivetrain(MecanumDrivetrainConstants constants) {
>   return addDrivetrain(new MecanumDrivetrain(constants));
> }

Okay, this sounds good.

closed 2025-05-01 by BaronClaps