Pedro-Pathing/PedroPathing #28: Split MathFunctions into multiple files

BeepBot99, 2025-04-30:
It is generally considered an antipattern to have a big "utils" file that contains all the logic. There is almost never a need for any utils file at all; it can just be contained in the DTOs themselves. For the utils that truly need to be in a utils file, it is best to split them into separate files.

Here are some examples:

- `nCr` does need to be in a utils file, as Java does not have extension methods. However, it might be better to put it in a dedicated `Combinatorics` class, for clarity, although `MathFunctions` is fine.
- `getSign` is not needed, there is already `Math.signum`
- `normalizeAngle` and other angle functions should probably be in an `Angle` class. This class would also allow different angle units and would be used everywhere an angle is used for type safety. Maybe a dedicated `math` package?
- `distance` can be `Pose.distance`, along with `addPoses`, `rotatePoses`, etc.
- `inToMm` and `mmToIn` can be in a `Distance` class.

The only methods that I really see need for a utils class are as follows:

- nCr
- clamp
- roughlyEquals

BaronClaps, 2025-05-01:
getSign has been removed and the Distance + Angle classes have been created, but are basically empty.

BaronClaps, 2025-05-01:
> * `distance` can be `Pose.distance`, along with `addPoses`, `rotatePoses`, etc.
`addPoses `+ `subtractPoses `would just be the minus and plus methods we already have? So we would only have to add `distance ` and `rotate`.

BeepBot99, 2025-05-01:
> `addPoses `+ `subtractPoses `would just be the minus and plus methods we already have? So we would only have to add `distance ` and `rotate`.

Yeah, and they should be instance methods. I'd call them distanceTo and rotate.

BaronClaps, 2025-06-27:
@BeepBot99 Do you still want to do this?

BaronClaps, 2025-07-05:
Commit c05c3db addresses this issue. Vector and Pose-related operations were moved into their respective classes. Distance conversions were removed.

closed 2025-07-05 by BaronClaps