Pedro-Pathing/PedroPathing #29: Implementation of Follower acting as a Orchestrator

BaronClaps, 2025-04-30:
Instead of the current method of the Follower performing calculations for the errors, vectors, and then powering the motors, the Follower will now simply act as an orchestrator of the Drivetrain, PoseUpdater, ErrorCalculator, and VectorCalculator, feeding the outputs of each one to the other, so that a final product of movement is achieved. 

- The PoseUpdater will handle the tracking of the Robot using a Localizer and produce many output, most importantly the current pose and current velocity.
- The ErrorCalculator will determine the errors between the currentPose and the closestPose based on the currentPath and velocity.
- The VectorCalculator will determine the vectors required to adjust the robot to align with the currentPath based on the errors produced by the ErrorCalculator.
- The Drivetrain will combine the vectors produced by the VectorCalculator to move/correct the robot along the currentPath. This will be done through various ways (determined by the user), but most notiably by 4 motors (for Mecanum), 4 motors and 4 servos (Coaxial Swerve), etc.

BeepBot99, 2025-04-30:
What would `VectorCalculator` look like?

BaronClaps, 2025-04-30:
> What would `VectorCalculator` look like?

It would take inputs of the errors and produce the corrective vectors

BeepBot99, 2025-04-30:
> > What would `VectorCalculator` look like?
> 
> It would take inputs of the errors and produce the corrective vectors

OK, so it would be the one with the PIDs, right?

BeepBot99, 2025-04-30:
> * The PoseUpdater will handle the tracking of the Robot using a Localizer and produce many output, most importantly the current pose and current velocity.

Possibly rename `PoseUpdater` to `PoseTracker`?

BaronClaps, 2025-05-01:
This should mostly be done - other than the Feedback manager. A few things need to happen with VectorCalculator to make the PIDFCoefficents accessible during tuning, but otherwise the Follower seems very close to being done. Around 500 lines in the follower now.

closed 2025-06-27 by BaronClaps