Pedro-Pathing/PedroPathing #31: Split Pedro into two modules

BeepBot99, 2025-04-30:
Split Pedro into two modules: `pedro` and `pedro-ftc`. This helps with:

- Decoupling Pedro from the FTC SDK
- Future-proofing Pedro to make the change to SystemCore in 2027 much easier
- Making it so when SystemCore is released, Pedro won't need to be either FTC SDK or WPILib, but both can be used and Pedro only needs to be updated in one place.

What would be in `pedro`:

- All the interfaces
- Follower
- Paths, Bézier curves, and Pose
- Utils

What would be in `pedro-ftc`:

- Implementations of `Drivetrain`, `Localizer`, and `Logger`

BaronClaps, 2025-05-01:
This would be interesting to explore before the 1.1.0 release, but let's focus on making sure the dev will work before doing this.

BeepBot99, 2025-05-01:
> This would be interesting to explore before the 1.1.0 release, but let's focus on making sure the dev will work before doing this.

Sounds good!

BaronClaps, 2025-06-27:
Let's come back to this later.