Pedro-Pathing/PedroPathing #35: Restructure Callbacks

Tyler-Stocks, 2025-05-06:
While there is a big restructure going on, I figured I'd throw my hat into the ring. 

Currently, all callbacks are handled through the PathCallbacks class located in the utils package.

There are two main issues with this approach
1. It is not super easy to add new callbacks (admittedly this is not a huge issue, but it could be better)
2. It is impossible for users to add there own callbacks

In order to address these issues I suggest having an abstract parent class PathCallback with concrete implementations for TemporalCallback, ParametricCallback, etc... . In my current prototype, the parent class looks like this: 

```java
public abstract class PathCallback extends SingleRunAction {
    protected final Follower follower;
    public final int index;

    public PathCallback(@NonNull Follower follower, @NonNull Runnable runnable, int index) {
        super(runnable);
        this.follower = follower;
        this.index = index;
    }

    public abstract boolean isReady();
}
```

Here are some advantages of this approach:
1. Easy to add more callbacks in the future. All that would be needed is a new child class and a method on 
    PathBuilder.
2. Users can add their own callbacks. I think the ability to define your own callback conditions is something that could be very useful to lots of people.
3. Simpler code in the follower. The code in the follower would now be:
    ```java
    public void updateCallbacks() {
        for (PathCallback callback : currentPathChain.getCallbacks()) {
            if (!callback.isReady() || callback.hasBeenRun()) return;
            callback.run();
        }
    }
    ```

The one major issue I have with this design is increased coupling. This implementation of PathCallback requires the Follower to be passed to both PathBuilder and PathCallback which isn't ideal. 
Personally, I think the advantages outweigh the increased coupling but I'd love to hear any suggestions/feedback

junkjunk123, 2025-05-07:
> While there is a big restructure going on, I figured I'd throw my hat into the ring.
> 
> Currently, all callbacks are handled through the PathCallbacks class located in the utils package.
> 
> There are two main issues with this approach
> 
> 1. It is not super easy to add new callbacks (admittedly this is not a huge issue, but it could be better)
> 2. It is impossible for users to add there own callbacks
> 
> In order to address these issues I suggest having an abstract parent class PathCallback with concrete implementations for TemporalCallback, ParametricCallback, etc... . In my current prototype, the parent class looks like this:
> 
> public abstract class PathCallback extends SingleRunAction {
>     protected final Follower follower;
>     public final int index;
> 
>     public PathCallback(@NonNull Follower follower, @NonNull Runnable runnable, int index) {
>         super(runnable);
>         this.follower = follower;
>         this.index = index;
>     }
> 
>     public abstract boolean isReady();
> }
> Here are some advantages of this approach:
> 
> 1. Easy to add more callbacks in the future. All that would be needed is a new child class and a method on PathBuilder.
> 2. Users can add their own callbacks. I think the ability to define your own callback conditions is something that could be very useful to lots of people.
> 3. Simpler code in the follower. The code in the follower would now be:
>    public void updateCallbacks() {
>        for (PathCallback callback : currentPathChain.getCallbacks()) {
>            if (!callback.isReady() || callback.hasBeenRun()) return;
>            callback.run();
>        }
>    }
> 
> The one major issue I have with this design is increased coupling. This implementation of PathCallback requires the Follower to be passed to both PathBuilder and PathCallback which isn't ideal. Personally, I think the advantages outweigh the increased coupling but I'd love to hear any suggestions/feedback

I was wondering, why is the follower needed in the PathCallback class? And why does index need to be included as well?

BeepBot99, 2025-05-07:
Maybe something like this?
```java
public interface PathCallback {
  void run(double pathIndex);
  boolean isReady();
  default void reset();
}

public class SingleRunCallback implements PathCallback {
  private final PathCallback callback;
  private boolean hasRun = false;

  public SingleRunCallback(PathCallback callback) {
    this.callback = callback;
  }

  @Override
  public boolean isReady() {
    boolean isReady = callback.isReady() && !hasRun;
    hasRun = true;
    return isReady;
  }

  @Override
  public void reset() {
    hasRun = false;
  }

  @Override
  public void run() {
    callback.run();
  }
}
```

(this implements the decorator pattern btw)

junkjunk123, 2025-05-07:
Yeah lol I figured this out. I actually just sent out a push with a restructure about 2 hours ago, would you like to take a look?

BeepBot99, 2025-05-07:
> > I was wondering, why is the follower needed in the PathCallback class?
> 
> Good question! 
> 
> Callbacks need to know the index of the path in the running path chain to determine whether it is time for them to run or not. Additionally, specific callbacks such as temporal callback and parametric callback need access to information the follower provides (path start times, current path progress, etc. )
> 
> One way we could avoid this is to take a similar approach to the way it is currently implemented. That is, doing to processing of when a callback should be run in the Follower. 

Or have the ones that need follower use constructor injection instead

Tyler-Stocks, 2025-05-07:
> public interface PathCallback {
>   void run();
>   boolean isReady();
>   default void reset();
> }

I like this interface, but I'm not sure we even need run or reset as they are handled by the single run action. 
Keeping reset might be a good idea though as I could see some custom logic going there. 

A concrete implementation could look something like this:

```java
public class TemporalCallback extends SingleRunAction implements PathCallback {
    private final long milliseconds;
    private final Follower follower;
    private final int index;

    public TemporalCallback(
            @NonNull Follower follower,
            @NonNull Runnable runnable,
            long milliseconds,
            int index
    ) {
        super(runnable);
        this.milliseconds = Math.max(0, milliseconds);
        this.follower = follower;
        this.index = index;
    }

    @Override public boolean isReady() {
        return System.currentTimeMillis() - follower.getPathStartTimes()[index] > milliseconds;
    }
}
```

I was also thinking we might want to abstract the general "check if we are at the right path index" to the follower. So have something like this

```java
public void updateCallbacks() {
    for (PathCallback callback : currentPathChain.getCallbacks()) {
        if (callback.index() != getChainIndex()) continue;
        if (!callback.isReady()) continue;
        if (callback.hasBeenRun()) continue;
        callback.run();
    }
}
```

Tyler-Stocks, 2025-05-07:
> Yeah lol I figured this out. I actually just sent out a push with a restructure about 2 hours ago, would you like to take a look?

Sure!

Tyler-Stocks, 2025-05-07:
> Yeah lol I figured this out. I actually just sent out a push with a restructure about 2 hours ago, would you like to take a look?

It looks good to me.

BeepBot99, 2025-05-07:
I think it would be better to use the decorator pattern, so instead of the callbacks extending single run action, single run action takes a callback as a constructor parameter, and the callback does not know anything about single run action. Single run action adds behavior to the callback without the callback knowing.

BeepBot99, 2025-05-07:
Also one small thing, you should use nanoTime instead of currentTimeMillis for the callback, as you want to use a monotonic clock and not a wall clock.

Tyler-Stocks, 2025-05-07:
Would it make more sense to use the NanoTimer present in the util class for Temporal callback?

Tyler-Stocks, 2025-05-07:
I agree, but I figured we might as well use it if we have it. And it's beyond the scope of this issue

BeepBot99, 2025-05-07:
> I agree, but I figured we might as well use it if we have it. And it's beyond the scope of this issue

I disagree. I think it's so simple and it's built into both the Java stdlib and the FTC SDK that it should not be used anyway.

Tyler-Stocks, 2025-05-08:
We should probably make a seperate issue for this

BaronClaps, 2025-05-08:
Havish handled the callbacks 👍🏻 

https://github.com/Pedro-Pathing/PedroPathing/commit/1ea6d558f3222ece7fdf0077f8b7478f80b8366e

closed 2025-05-08 by BaronClaps