Pedro-Pathing/PedroPathing #42: Make Follower#pathBuilder static

BeepBot99, 2025-06-16:
The constructor of `PathBuilder` takes a `Follower` and, optionally, a `PathConstraints`. Unless I am missing something, it does not appear that the `Follower` is used anywhere. As for `PathConstraints`, it should default to `PathConstraints.defaultConstraints`. With those two changes, it does not need any instance data of `Follower` anymore. Then it can be made into a static method instead. This also means that there is no need for a `buildPaths` method in an OpMode anymore, because it can just be done using field initializers:

```java
private PathChain toBasket = Follower.pathBuilder()
                                     .something()
                                     .build();
```

BaronClaps, 2025-06-23:
You can do 

```java
private PathChain toBasket = new PathBuilder()
                                     .something()
                                     .build();
```

BeepBot99, 2025-06-23:
Sure, but then why have `follower.pathBuilder()` at all? Also though, it is common to have a static method to create a builder, and in my opinion `Follower.pathBuilder()` looks better.

closed 2025-06-27 by BaronClaps