Pedro-Pathing/PedroPathing #64: Drive vector pointing the wrong direction at parametric end

monkbroc, 2025-09-30:
We're running into an issue visible in the Line test and DriveTuner where the drive vector makes the robot move away from the last point after reaching the parametric end of the path. We did PID tuning and while not perfect, it's decent.

Here's a trace showing the problem
[image]

I'm seeing that in `getDriveError()` when `currentPath.isAtParametricEnd()`, it calculates the distanceToGoal from the current pose and the current path last control point.
https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/ErrorCalculator.java#L193-L195

If my current path last control point is 40, 0 and my current pose is 41, 0, the offset in `getDriveError()` is 1, 0. The current path end tangent is magnitude 1, theta 0. Therefore the distance to goal is 1 inch. However it's 1 inch backwards, not forward.

Then `driveError = getDriveVelocityError(distanceToGoal);` gets called. In there `distanceToGoalVector` is set to the closet point tangent (magnitude 1, theta 0) times `distanceToGoal`. This vector points forward, away from the current path last control point. `forwardDistanceToGoal` and `forwardVelocityGoal` are also positive numbers. This makes the robot accelerate away from the goal.

This behavior goes away completely when setting `timeoutConstraint = 0` in PathConstraints showing that this is due to the behavior at parametric end, not PID tuning overshooting.

[image]

Have you seen this before? Is it possible we have a constant with the wrong sign? I don't see a constant that would change this behavior.

I suspect that changing the sign on this line in `getDriveError` would fix it. What do you think?
```
offset.setOrthogonalComponents(currentPath.getLastControlPoint().getX() - currentPose.getX(), currentPath.getLastControlPoint().getY() - currentPose.getY());
```

monkbroc, 2025-10-01:
I gave it a test with the opposite sign on `offset`, but that just introduced strong oscillations at the parametric end.

I traced through the code for Pedro Pathing 1.0.9 and it looks the same. The only difference I found was `(forwardDistanceToGoal <= 0 ? 1 : -1)` in the old code but not the new, but it's inside an `abs()` so I don't think it matters:
```
double forwardVelocityGoal = MathFunctions.getSign(forwardDistanceToGoal) * Math.sqrt(Math.abs(-2 * currentPath.getZeroPowerAccelerationMultiplier() * forwardZeroPowerAcceleration * (forwardDistanceToGoal <= 0 ? 1 : -1) * forwardDistanceToGoal));

    public static double getVelocityToStopWithDeceleration(
            double directionalDistance,
            double deceleration
    ) {
        return Math.signum(directionalDistance)
                * Math.sqrt(Math.abs(-2 * deceleration * directionalDistance));
    }
```

Do you have any idea what's going on here?

monkbroc, 2025-10-01:
Hi Havish. Thanks for the quick answer and thanks for your contributions to Pedro Pathing. It's a great resource for all FTC teams!

The drive vector is still used when the robot is at the parametric end of the path until the path end constraints are satisfied.

To show this, I did a test by editing `getDriveError()` to make it possible to return 0 when at parametric end.

```
    public boolean zeroDriveAtParametricEnd = false;
    public double getDriveError() {
        // no change here...
        if (!currentPath.isAtParametricEnd()) {
            // no change here...
        } else {
            if (zeroDriveAtParametricEnd) {
                return 0;
            }
            Vector offset = new Vector();
            offset.setOrthogonalComponents(currentPose.getX() - currentPath.getLastControlPoint().getX(), currentPose.getY() - currentPath.getLastControlPoint().getY());
            distanceToGoal = currentPath.getEndTangent().dot(offset);
        }

        driveError = getDriveVelocityError(distanceToGoal);

        return driveError;
    }
```

Here's a trace with the first 2 back and forth movements with `zeroDriveAtParametricEnd = false` showing the issue I'm describing (driving away from the end of the path) and the next 2 with `zeroDriveAtParametricEnd = true` showing no issue.
[image]

Here's the path I see in the code in this situation:
* The op mode calls [Follower.update](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L385).
* We're not in manual drive, there's a current path, we're not holding position, and we're busy since the path end constraints are not reached so we go in this [if (busy)](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L418) condition.
* `updateErrorAndVectors` is [called in the if condition](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L422).
* The vectors are updated in [updateVectors](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L375) and the drive error is set to `useDrive && !holdingPosition ? getDriveError() : -1` which is positive if the robot is at parametric end past the last point due to [this calculation](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/ErrorCalculator.java#L193-L195).
* The drivetrain is updated with `drivetrain.runDrive(getCorrectiveVector(), getHeadingVector(), getDriveVector(), poseTracker.getPose().getHeading());`
* Since `getDriveVector()` is facing in the tangent of the last point, the robot drives away from the last point.
* The function `update()` continues past [this if](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L432) since we're at parametric end.
* `updateErrorAndVectors` is [called again in the same execution of the update() function](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L461) which seems undesirable. Is this a separate bug?
* [The path end constraints are evaluated](https://github.com/Pedro-Pathing/PedroPathing/blob/main/core/src/main/java/com/pedropathing/follower/Follower.java#L462-L481).
* If the constraints are satisfied, it will either start holding or break following. In the case I'm showing here, since the robot is driving away from the last point, the `poseTracker.getPose().distanceFrom(closestPose.getPose())` constraint will remain false so the path end constraint that triggers is always the timeout.

Here's our constants file:
```
public class Constants {

    public static FollowerConstants followerConstants = new FollowerConstants()
            .mass(5.816)
            .forwardZeroPowerAcceleration(-52.1)
            .lateralZeroPowerAcceleration(-62.1)
            .useSecondaryTranslationalPIDF(false)
            .useSecondaryHeadingPIDF(false)
            .useSecondaryDrivePIDF(false)
            .centripetalScaling(0.0005)
            .translationalPIDFCoefficients(new PIDFCoefficients(0.04, 0, 0, 0.015))
            .headingPIDFCoefficients(new PIDFCoefficients(0.5, 0, 0.01, 0.015))
            .drivePIDFCoefficients(new FilteredPIDFCoefficients(0.01, 0, 0.001, 0.6, 0.015));

    public static MecanumConstants driveConstants = new MecanumConstants()
            .leftFrontMotorName("frontLeft")
            .leftRearMotorName("backLeft")
            .rightFrontMotorName("frontRight")
            .rightRearMotorName("backRight")
            .xVelocity(66.75)
            .yVelocity(54.1)
            .leftFrontMotorDirection(DcMotorSimple.Direction.FORWARD)
            .leftRearMotorDirection(DcMotorSimple.Direction.FORWARD)
            .rightFrontMotorDirection(DcMotorSimple.Direction.REVERSE)
            .rightRearMotorDirection(DcMotorSimple.Direction.REVERSE);

[... 2 lines cut ...]
            .strafePodX(-81.19)
            .distanceUnit(DistanceUnit.MM)
            .hardwareMapName("odo")
            .yawScalar(1.0)
            .encoderResolution(
                    GoBildaPinpointDriver.GoBildaOdometryPods.goBILDA_4_BAR_POD
            )
            .forwardEncoderDirection(GoBildaPinpointDriver.EncoderDirection.REVERSED)
            .strafeEncoderDirection(GoBildaPinpointDriver.EncoderDirection.FORWARD);

    public static PathConstraints pathConstraints = new PathConstraints(
            0.995,
            500,
            1.2,
            1
    );
    public static Follower createFollower(HardwareMap hardwareMap) {
        return new FollowerBuilder(followerConstants, hardwareMap)
                .mecanumDrivetrain(driveConstants)
                .pinpointLocalizer(localizerConstants)
                .pathConstraints(pathConstraints)
                .build();
    }
}
```

BeepBot99, 2025-10-01:
@monkbroc Just curious, what software are you using for those graphs?

monkbroc, 2025-10-01:
> Did you intend to put your yawScalar at 1.0?

We did not set this intentionally. We we using the [PedroPathing 1.0.9 Pinpoint config](https://github.com/Pedro-Pathing/Documentation-1.0.9/blob/5c57f585b72d16222611d656f6debe266d1d337a/pedro/localization/pinpoint.md?plain=1#L17) and yawScalar got converted over by the [constants migrator](https://pedro-pathing.github.io/ConstantsConverter/) despite LConstants having `useYawScalar = false`. I had removed the customEncoderResolution but hadn't touched yawScalar. Good to know it's best to remove it.

[image]

I tried to address the issue of the constants converter ignoring `useYawScalar` and `useCustomEncoderResolution` in [this PR](https://github.com/Pedro-Pathing/ConstantsConverter/pull/1), but I pushed it right as a major refactor of the tool got pushed so I ended up closing the PR. I can resubmit that PR if it's helpful.

For graphs, I use the Logging widget in FTC Dashboard to download a CSV of all the telemetry variables and use [Flow CSV Viewer](https://apps.microsoft.com/detail/9nq7z06vrxbw) to plot the data. It works pretty well to dig deep into issues. I haven't figured out if Panels can do this as its telemetry seems to be all strings.

[image]

Here's the telemetry block I added to the test op mode:
```
        Telemetry dashTelemetry = FtcDashboard.getInstance().getTelemetry();
        dashTelemetry.addData("isBusy", follower.isBusy());
        dashTelemetry.addData("atParametricEnd", follower.atParametricEnd());
        dashTelemetry.addData("distanceTraveledOnPath", follower.getDistanceTraveledOnPath());
        dashTelemetry.addData("pathCompletion", follower.getPathCompletion());
        dashTelemetry.addData("distanceRemaining", follower.getDistanceRemaining());
        dashTelemetry.addData("tValue", follower.getCurrentTValue());
        dashTelemetry.addData("x", follower.getPose().getX());
        dashTelemetry.addData("y", follower.getPose().getY());
        dashTelemetry.addData("h", follower.getPose().getHeading());
        dashTelemetry.addData("closestX", follower.getClosestPose().getPose().getX());
        dashTelemetry.addData("closestY", follower.getClosestPose().getPose().getY());
        dashTelemetry.addData("distanceClosest", follower.getPose().distanceFrom(follower.getClosestPose().getPose()));

        dashTelemetry.addData("velocityM", follower.getVelocity().getMagnitude());
        dashTelemetry.addData("velocityT", follower.getVelocity().getTheta());
        dashTelemetry.addData("accelerationM", follower.getAcceleration().getTheta());
        dashTelemetry.addData("accelerationT", follower.getAcceleration().getTheta());

        dashTelemetry.addData("driveError", follower.getDriveError());
        dashTelemetry.addData("headingError", follower.getHeadingError());
        dashTelemetry.addData("translationalErrorM", follower.getTranslationalError().getMagnitude());
        dashTelemetry.addData("translationalErrorT", follower.getTranslationalError().getTheta());

        dashTelemetry.addData("driveVectorM", follower.getDriveVector().getMagnitude());
        dashTelemetry.addData("driveVectorT", follower.getDriveVector().getTheta());
        dashTelemetry.addData("headingVectorM", follower.getHeadingVector().getMagnitude());
        dashTelemetry.addData("headingVectorT", follower.getHeadingVector().getTheta());
        dashTelemetry.addData("translationalCorrectionM", follower.getTranslationalCorrection().getMagnitude());
        dashTelemetry.addData("translationalCorrectionT", follower.getTranslationalCorrection().getTheta());
        dashTelemetry.addData("centripetalForceCorrectionM", follower.getCentripetalForceCorrection().getMagnitude());
        dashTelemetry.addData("centripetalForceCorrectionT", follower.getCentripetalForceCorrection().getTheta());
        dashTelemetry.addData("correctiveVectorM", follower.vectorCalculator.getCorrectiveVector().getMagnitude());
        dashTelemetry.addData("correctiveVectorT", follower.vectorCalculator.getCorrectiveVector().getTheta());
        dashTelemetry.update();
```

monkbroc, 2025-10-01:
> I believe it is now fixed. If you could try converting again, that would be great!

The converter screenshot is from today. If you put the values from the [1.0.9 docs](https://github.com/Pedro-Pathing/Documentation-1.0.9/blob/5c57f585b72d16222611d656f6debe266d1d337a/pedro/localization/pinpoint.md?plain=1#L17) in to the converter
```
    PinpointConstants.useYawScalar = false;
    PinpointConstants.yawScalar = 1.0;
    PinpointConstants.useCustomEncoderResolution = false;
    PinpointConstants.encoderResolution = GoBildaPinpointDriver.GoBildaOdometryPods.goBILDA_4_BAR_POD;
    PinpointConstants.customEncoderResolution = 13.26291192;
```
you'll get this in the output
```
    .yawScalar(1.0)
    .encoderResolution(
      GoBildaPinpointDriver.GoBildaOdometryPods.goBILDA_4_BAR_POD
    )
    .customEncoderResolution(13.26291192)
```

The converter doesn't take into account `useYawScalar` and `useCustomEncoderResolution`.

> Along with that, can you try the new default [pathConstraints](https://github.com/Pedro-Pathing/Quickstart/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/pedroPathing/Constants.java#L12)? The timeoutConstraint is 100.

Sure. That does reduce the impact of the issue.

Note that the converter still outputs these path constraints:
```
  public static PathConstraints pathConstraints = new PathConstraints(
    0.995,
    500,
    1,
    1
  );
```

BeepBot99, 2025-10-01:
> This is true, but now there is no boolean to set if you want to use either two, it is now an optional double, so if a value is provided, it will use that value.

Oh I was talking about `PathConstraints` not `yawScalar`. I can fix `yawScalar`.

closed 2025-10-17 by BaronClaps

monkbroc, 2025-10-17:
Thanks @BaronClaps. We'll test this fix over the weekend and I'll give a 👍 to confirm that it solves the issue.

monkbroc, 2025-10-18:
Works great! Now if the robot goes past the end point, it corrects backwards and stops very close to the target. The path timeout is still 500 ms in the trace below to illustrate the fix, but for our real op mode, we'll use 100 ms.

[image]